tscircuit/core #142: Shorten the value of resistance and capacitance

imrishabh18, 2024-10-07:
Here the Resistance should be `100k` and capacitance could be `0.1uF` 

[image]

imrishabh18, 2024-10-08:
@seveibar Currently, what I am doing in `circuit-to-svg` is looping through the circuit-json again to match the `source_component_id` and then get the name and value. A better way would be having these value in the `schematic_component` itself, we can reduce that extra loop to get the value What do you think?

[image]
[image]

seveibar, 2024-10-08:
The loop is fine! Note that if you use soup-util eventually it will automatically index to avoid looping

imrishabh18, 2024-10-08:
So, you mean to add this new `display_value` in the `source_component` itself?

closed 2024-12-23 by github-actions[bot]