tscircuit/core #221: Bug: `pcbRotation` not impacting output `cad_component` rotation on resistor

seveibar, 2024-11-01:
[image]

imrishabh18, 2024-11-04:
@seveibar The rotation currently rotates the copper pads of the board, and not the whole 3d-model. Take a look here

https://snippets.tscircuit.com/editor?&snippet_type=board#data:application/gzip;base64,H4sIAPmfKGcAA52UW2+bQBCF3/kVo30iUsDgGzQKUVWcVJWS1KLp7anCsMGr1ixa1m4rxH/vAIsBX5Sob+zZM9+cYRfYJuNCQgHbnAY0vqcxlPAs+AbIW5lHbJTTHWWrUJiCxsYvGhOtX/K0/ORM55a9XARfgvOViTHbGZIbk53hj92JM7WcISegOcslF5fVwg+zMGK4GgJFtGVyFHFBiabRP3VxxNNcwsNff80y8EDPBM/yK2Sm4YZeQS4FSxMoL8C7gUID5V9+DWx0H+bXSbVBLva+YN642nQ6CebkEumiFsI0wh7EfkDtmXOZYTOJgjW1xgSbdhzniOMcc8aW9fNFkntEck8kepHjq8n2r1onfjNavywKM8u1ZqjjU+Vr+db2bohTA/ZxzYT/hbu/Xajzaa6kTiqlPhhB5VakoOMjwPWKhyKG3yyWaw8xmw2BNWXJWqrVTW1DY3Ww6hkgi1bfvGJa9oWAy1AynnrkjRXThOz3bh89klJp7mY9bdloSYqfQyt+OGH8+PlJiZMfk1Ye7VPh/WqyOOUgQ6FCoMpS2yuq9Cayyn6tM6j97hW2ORsY3OFctlszD6uMierS5exB8J5kPN8PBilNutGHW72iOtq5zhiy6Vw1RkARzE0MMC4P4B2vvg7tofViH4cpAlexsLotlyKMaP0jUW/y7l0Jkvcad63Oe53Ka5/zNtuIQu+pEY647x8XrzL7joljvtaK76CzDs7yelR/K9XiQiu1fxbIQPf3BQAA

So, what is the expected outcome when we have `group` and not `board` on rotation?

seveibar, 2024-11-04:
The 3d model is supposed to rotate to match the pads.

closed 2024-11-21 by Abse2001