tscircuit/core #248: Figure out why schematic layout and circuit-to-svg are so big, reduce their size and update core

seveibar, 2024-11-07:
These modules have dramatically increase core size, it looks like it's probably files being accidentally bundled, we should make sure they use `tsup-node` (not `tsup` which bundles dependencies) have have the `package.json` `"files"` defined properly

[image]

/bounty $15

akhilender-bongirwar, 2024-11-08:
> [image]

@seveibar, is there a way to generate this report locally to test changes ?

seveibar, 2024-11-08:
@akhilender-bongirwar the package.json file should have a command that uses iirc `howfat`

akhilender-bongirwar, 2024-11-09:
@seveibar, I actually tried `bun run measure-bundle` and encountered several warnings, causing the terminal to crash. This happened twice, which is why I reached out here.

[image]

> the package.json file should have a command that uses iirc howfat

However, after seeing your reply, I cloned the repository again and tried once more. This time, despite the warnings, the table was generated.

> you can always take a peek at the workflow that writes that comment as well inside the `.github/workflows` directory

The changes I made didn’t reduce the size, so I decided not to proceed under the issue.

Thank you for your response : )

closed 2024-11-11 by imrishabh18