tscircuit/core #265: Fix the normalizer function logic

imrishabh18, 2024-11-10:
Test file - `chip-pin-alias.test.tsx` 

This test should _not_ rely on snapshots however!!!! Load in all the `source_port`s, sort them, and confirm in code that they match expectations. 

FWIW i think this test in particular points to a very confusing behavior from the normalizer. `B11` and `B112` should probably be `B1_1` and `B1_2`!!! We can fix it in a future PR and look for these edge cases with the normalizer- but really this test reproduces a bug and we should leave a comment in the test that indicates that it's a known issue that should be fixed

_Originally posted by @seveibar in https://github.com/tscircuit/core/pull/263#discussion_r1835569101_

anas-sarkez, 2024-11-22:
@imrishabh18  this issue needs to close
  [image]

imrishabh18, 2024-11-22:
Thanks for pointing out!

closed 2024-11-22 by imrishabh18