tscircuit/core #307: Don't wire through manufacturer labels in schematic

seveibar, 2024-11-15:
This requires moving the manufacturer labels as a `schematic_text` core output, rather than inserting inside of `circuit-to-svg`

[image]

/bounty $30

anas-sarkez, 2024-11-17:
@seveibar 
I've created a function that handles the chip manufacturer's label and chip name, and converts them into a `schematic-text` element. How can I modify the trace so it does not get passed to the `schematic-text`?
[image]

seveibar, 2024-11-17:
Inside `Trace.ts` `schematic_text` will need to be added as an obstacle

anas-sarkez, 2024-11-17:
@seveibar 
this is after added `schematic_text` to obstacle
[image]

seveibar, 2024-11-17:
nice that looks correct, the trace is no longer going through the schematic trace

seveibar, 2024-11-17:
we should also make it not go through the symbol 🤔

anas-sarkez, 2024-11-17:
> we should also make it not go through the symbol 🤔 (probably a separate PR i guess)

[image]
It's already added to obstacle

PR #322 (2024-11-17): Refactor the trace so it does not get passed to the chip manufacturer part number and name
this issue fixed with converting manufacturer labels as a `schematic_text`
/claim #307
fixed #307 
**this example before fixed the issue**

[image]

**this is after fixed** 
[image]
closed 2024-11-17 by anas-sarkez