tscircuit/core #331: Support Rotated Chips

seveibar, 2024-11-19:
- [ ] Requires: https://github.com/tscircuit/circuit-json/issues/66
- [ ] Make sure that https://github.com/tscircuit/autorouting knows how to turn `rotated_rect` pads into an obstacle
- [ ] Make sure `pcb-viewer` knows how to draw rotated rects

/bounty $65

ShiboSoftwareDev, 2024-11-19:
@seveibar what about a rotation prop for smtpads?

ShiboSoftwareDev, 2024-11-19:
/attempt #331

ShiboSoftwareDev, 2024-11-21:
@seveibar what do you mean with the fourth point?

seveibar, 2024-11-21:
@ShiboSoftwareDev yea not written well. That bullet point is basically saying "if a chip is rotated 45 degrees, make sure its smtpads all have the proper rotation

ShiboSoftwareDev, 2024-11-21:
> [@ShiboSoftwareDev](https://github.com/ShiboSoftwareDev) yea not written well. That bullet point is basically saying "if a chip is rotated 45 degrees, make sure its smtpads all have the proper rotation

basically inherit rotation from chip

ShiboSoftwareDev, 2024-11-28:
@seveibar how can I test the rotated rect on the autorouting repo? Also there are multiple autorouters, do I have to implement it on all of them seperately or try to create a global function, I haven't worked on that repo before and it's not super clear.

seveibar, 2024-11-28:
@ShiboSoftwareDev theres a function called "getObstaclesFromSoup", i think the main goal is to create eg 3-5 rectangles approximating the rotated rect, that will fix all the autorouters

seveibar, 2024-11-28:
@ShiboSoftwareDev this is an example of an approximation, the formula is incorrect however (doesn't properly cover at 80 degrees)

[image]

```ts
const generateApproximatingRects = () => {
    const rects = [];
    const sliceWidth = width / numRects;
    
    for (let i = 0; i < numRects; i++) {
      const x = (i - numRects/2) * sliceWidth + sliceWidth/2;
      
      // Calculate offset based on rotation
      const rotatedX = x * cosAngle;
      const rotatedY = x * sinAngle;
      
      // Calculate the maximum height needed for full coverage at this slice
      // This is the sum of the vertical projections of both width and height
      const verticalProjectionFromWidth = Math.abs(sliceWidth * sinAngle);
      const verticalProjectionFromHeight = Math.abs(height * cosAngle);
      const sliceHeight = verticalProjectionFromWidth + verticalProjectionFromHeight;
      
      rects.push({
        x: centerX + rotatedX - sliceWidth/2,
        y: centerY + rotatedY - sliceHeight/2,
        width: sliceWidth,
        height: sliceHeight
      });
    }
    
    return rects;
  };
```

seveibar, 2024-11-28:
For [tscircuit/checks](https://github.com/tscircuit/checks) we'll want to implement a "true collision check" rather than an approximation, but it's so much easier/faster to have upright rects in the autorouter in the short term that I think we should settle for the approximation. Future versions of the autorouter will do it properly. Note that we also approximate plated holes as rects right now

PR #401 (2024-12-10): rotated rect obstacle test
/claim #331
/closes #331
closed 2024-12-10 by ShiboSoftwareDev