tscircuit/core #345: Error:  Cannot read properties of undefined reading getPortSelector

seveibar, 2024-11-23:
[Snippet code to reproduce](https://tscircuit.com/editor?#data:application/gzip;base64,H4sIAD1pNmcAA32PQUvEMBCF7/kVQ07todtVRGQxi7iCICiiFPTYZuNasEmYSUQI+e+mzVrqQWEOM7zvvZnpB2vQQQBPqqFuBxHe0AzArxzJvib1qfquxRUN+8pTV0nOlpanx9P12fovE9pR5Yypr8kijSYH2XNv9v5DgYDCorG0SXm6HdQGyGGvDxBLEFsIDH65Ej9vLXhzwssZaJ6vd1ke/yj4XRZToXIeNRSpBbg8oPF2O/VpOsZa2b2KcBGhnpUpj+T7iwjVeTwSVUZ+GIetVNPvIoz86vbhJoIzgmvlxoEvApdwXvsfno+t52tLFr8B4ecCa6wBAAA=)

### Error
```
Render Error: Cannot read properties of undefined (reading 'getPortSelector')
```

seveibar, 2024-11-23:
/bounty $15

MustafaMulla29, 2025-02-08:
@seveibar can you please explain me how do i test this and run the "core" repo?

MustafaMulla29, 2025-02-08:
/attempt #345

MustafaMulla29, 2025-02-08:
@seveibar Hello, I have submitted a PR with proof that the issue has been solved. Can you please review it?

MustafaMulla29, 2025-02-08:
@seveibar Hello, I have made a PR with proof which solves the issue, can you please review this: https://github.com/tscircuit/core/pull/607

MustafaMulla29, 2025-02-18:
@seveibar is this error message : Received message: "undefined is not an object (evaluating 'v.getPortSelector')"
considered same as : Cannot read properties of undefined (reading 'getPortSelector')

"Received message: "undefined is not an object (evaluating 'v.getPortSelector')""  this is the error message when I test it using an undefined port.

Anshgrover23, 2025-03-23:
/attempt #345

closed 2025-03-23 by imrishabh18