tscircuit/core #348: Implement "render function timeouts" to prevent a specific render function e.g. `doInitial*` from running for more than a configurable 500ms

seveibar, 2024-11-23:


ShiboSoftwareDev, 2024-11-24:
/attempt #348 I'll give it a shot

seveibar, 2024-11-24:
@ShiboSoftwareDev I've been looking into this one for a bit, I think it's basically impossible w/o webworkers, I probably should have investigated more before bountying. Lmk if you come to a different conclusion!!!

ShiboSoftwareDev, 2024-11-25:
> [@ShiboSoftwareDev](https://github.com/ShiboSoftwareDev) I've been looking into this one for a bit, I think it's basically impossible w/o webworkers, I probably should have investigated more before bountying. Lmk if you come to a different conclusion!!!

@anas-sarkez

seveibar, 2024-11-30:
At this point I think this is mostly not possible (technically there is probably a solution but it's a dramatic architecture change), there is a version of this that is possible by considering the maximum execution time of effects, but imo we should basically do most of the freeze detection via webworkers

closed 2025-02-14 by github-actions[bot]