tscircuit/core #359: Emit events and the start and end of rendering for every `Renderable`, send to root circuit

seveibar, 2024-11-27:
Two main events:

- `renderable:renderLifecycle:<phase name>:start`
- `renderable:renderLifecycle:<phase name>:end`

e.g.

- `renderable:renderLifecycle:doInitialPcbComponentRender:start`
- `renderable:renderLifecycle:doInitialPcbComponentRender:end`

(the root circuit is the circuit defined in `Circuit.ts`, commonly used as `this.root`, all events from renderable are re-emitted)

The event payloads should include:

```js
{ renderId }
```

Create the utility event `"renderable:renderLifecycle:anyEvent"` to make it easier to listen to renderable events.

Make sure to test by recording all the events that are emitted using `circuit.on("renderable:anyEvent")

/bounty $30

seveibar, 2024-11-27:
Looking good @anas-sarkez!!!!

anas-sarkez, 2024-11-27:
[image]

anas-sarkez, 2024-11-27:
/attempt #359

PR #362 (2024-11-27): Implemented new event `renderable:renderLifecycle:anyEvent`  to make it easier to listen to renderable events
/claim #359
/closes #359

**example of how it works** 

[image]
closed 2024-11-27 by seveibar