tscircuit/core #380: The pcbX and pcbY when having a negative value as a string prop defaults to 0

imrishabh18, 2024-12-03:
https://snippets.tscircuit.com/editor?&snippet_type=board#data:application/gzip;base64,H4sIAHbzTmcAA52SMWvDMBCFd/+KQ1MypLY8W546dO7UUjoo9qUS2JaQLzQg8t+jWEqwSQght4h793S8DwkP1jiCFndy3xGs1iBqWGUA1dZI18K/bkkJxou+Z6BQ/ylKXR1MweZw1CMZN3UAg+xRsE/OUh/HcmhQeF4ck7ozhqzTAwmflLDpKsbNSbWdJGyV6XCmApxTfwTvKPwPs3rg7PeyPBma7ZfwG36jfgt/zRFrVNKG0I12TYeX4LHMntC96wAVzkDwVi6vnnPNxvNh/gpGeReDbcplrAmDFUvxeYzyMcZiOMOo8psnis7JU+XTj6mzdXYC9AmlvlUCAAA=

As you can see in this snippet that when you pass a negative value (pcbX) as a string prop to the first `smtpad` of the footprint it defaults to 0. But works fine for the the second smtpad.

imrishabh18, 2024-12-03:
/bounty $5

UdaykiranRegimudi, 2024-12-04:
/attempt #380

PR #589 (2025-01-30): fix: parse negative value pcb props
/close #380
closed 2025-01-30 by imrishabh18