tscircuit/core #387: Automatic labeling for passives

seveibar, 2024-12-05:
We have a trace with a passive on one side and a chip on the other side we it's eligible to be automatically turned into a net label style schematic trace it will be automatically converted if there are more than five junctions five crossings or five turns

AnasSarkiz, 2024-12-05:
/attempt #387

AnasSarkiz, 2024-12-07:
[image]

seveibar, 2024-12-07:
That looks right!!! @AnasSarkiz (nice!!!)

Can i make a small adjustment, can we make the netlabels have the refdes of the chip? So "U1_OUT1" instead of "OUT1"? Should help prevent "collisions"

Nicely done!

AnasSarkiz, 2024-12-07:
@seveibar 

[image]

seveibar, 2024-12-07:
Looks good although im surprised some of the long traces for those resistors arent converting (maybe we need to tweak conditions? Maybe 4 turns should convert?

AnasSarkiz, 2024-12-07:
How about a chip with four traces in the same face, and each trace has more than four turns? should we replace all of them with a netlabel? because I thought that we must have more than five traces on each face

seveibar, 2024-12-07:
@AnasSarkiz lets not consider faces yet, im looking a bit closer and i think even traces with more than 5 turns arent getting converted right now (am i wrong)

Im not opposed to the face thing, but if we can keep it simple thats better

seveibar, 2024-12-07:
Can i see it reduced to 4 turns? Maybe after we get that working we could do the face logic as an extension? Curious about the code complexity

AnasSarkiz, 2024-12-07:
@seveibar I modified the logic if one of the condition get true `complex trace's index is >= 5 for on the same face` **or** `the trace has _countComplexElements >=4`  

Output

[image]

[image]

[image]

[image]

seveibar, 2024-12-07:
I think of we were to do an approach with sides, we would maybe decrease the allowed trace complexity of ours on a heavily populated side, but because of the relative complexity of that the customization parameters couldn't be as simple, so better to keep it simple with a single parameter for now

AnasSarkiz, 2024-12-07:
it's my bad. I thought that we should have more than five complex traces on each side of a chip 
[image]

AnasSarkiz, 2024-12-07:
[image]
[image]

closed 2024-12-13 by AnasSarkiz