tscircuit/core #392: `pcb_port_id` is missing for `pcb_plated_hole` having shape `pill`

imrishabh18, 2024-12-06:
For this snippet - https://tscircuit.com/editor?snippet_id=99656a33-e266-466a-9870-395793f2d01e

Download the circuit-json and you will see the `pcb_plated_hole` having shape `pill` are not having the `pcb_port_id`.

imrishabh18, 2024-12-06:
/bounty $5

UTK-a-RSH, 2024-12-14:
/attempt #392

imrishabh18, 2024-12-14:
@seveibar I did more research on these, the problem why the `pcb_port_id` is missing is because the `pinLabels` are not present for the plated_holes `"alt_0", "alt_1"`, like it's present for others. So, I think this is not a bug and should rather throw error similar to when, the other labels are missing from pinLabels and used in `schPortArrangement`. Something like this 👇

[image]

Your view?

seveibar, 2024-12-15:
@imrishabh18 can you give an example of code that should throw?

imrishabh18, 2024-12-15:
@seveibar This is an example of code where the `platedhole` porthints are not in the pin labels and an error should be thrown .

```jsx
const pinLabels = {
    // 1: ["alt_2"],
    2: ["GND", "A1"]
  } as const

  circuit.add(
    <board width={10} height={10}>
      <chip
        name="test"
        cadModel={{
          objUrl:
            "https://modelcdn.tscircuit.com/easyeda_models/download?uuid=2a4bc2358b36497d9ab2a66ab6419ba3&pn=C165948",
          rotationOffset: { x: 0, y: 0, z: 180 },
          positionOffset: { x: 0, y: -2.5, z: 0 },
        }}
        pinLabels={pinLabels}
        supplierPartNumbers={{
          jlcpcb: ["C165948"],
        }}
        manufacturerPartNumber="TYPE-C-31-M-12"
        schPortArrangement={{
          leftSide: {
            pins: ["GND"],
            direction: "top-to-bottom",
          },
        }}
        footprint={
          <footprint>
            <platedhole
              portHints={["alt_2"]}
              pcbX="4.32511199999999mm"
              pcbY="-2.7740863499999477mm"
              outerHeight="1.7999964mm"
              outerWidth="1.1999975999999999mm"
              innerHeight="1.3999972mm"
              innerWidth="0.7999983999999999mm"
              shape="pill"
            />
           <smtpad
              portHints={["A1"]}
              pcbX="-3.3500060000000076mm"
              pcbY="2.449087049999889mm"
              width="0.29999939999999997mm"
              height="1.2999973999999999mm"
              shape="rect"
            />
          </footprint>
        }
      />
    </board>,
  )

```

seveibar, 2024-12-20:
@imrishabh18 because `// 1: ["alt_2"]` is commented out? If that's uncommented it should work right?

I don't think it's an error either way? `alt_2` would be assigned a new port number via our port numbering logic (probably `3`?)

It is a weird bit of code and it could warn or something but it's not obvious to me that it's an error

imrishabh18, 2024-12-21:
Yeah, if that is uncommented then all's good. There could be some warning when importing parts from jlcpcb, cause there are cases where the portHints are missing from the labels, which causes this situation to occur. Closing this issue, cause not a specific core issue failing to generate `pcb_port_id`.

closed 2024-12-21 by imrishabh18