tscircuit/core #426: Create <Resonator />

imrishabh18, 2024-12-16:
The resonator in our schematic symbol needs to be created with 3 pins. Take reference from how the `<Pinheader />` is implemented with multiple pins.

[image]

imrishabh18, 2024-12-16:
/bounty 10$

imrishabh18, 2024-12-16:
Yes! You will have to create the props in that repo

imrishabh18, 2024-12-16:
No need to create an issue for this in props. Just directly create a PR for it, and Crystal props should be similar to the resonator

imrishabh18, 2024-12-20:
Yes, you will have to add the `simple_resonator` in the source component of circuit json

Anshgrover23, 2024-12-20:
/attempt #426

closed 2024-12-20 by seveibar

Anshgrover23, 2024-12-21:
@techmannih you are feel free to make PR on this one, as i am still working it won't be an issue

PR #442 (2024-12-29): add: resonator
Fixes #426 
/claim #426

Note: please first merge this one: [implemented source_simple_resonator in tscircuit/circuit-json](https://github.com/tscircuit/circuit-json/pull/102)

- So that i can run tests and verify this one.