tscircuit/core #432: `<jumper />` should have option to have pins in multiple rows

imrishabh18, 2024-12-18:
For example the jumper has 6 pins, then there should be an option to split the `pcb_plated_holes` in two rows of 3. As you can see in the schematic below the pins are on both the sides, but for the footprint all the `plated_holes` are on a single line.

Schematic
[image]

Pcb
[image]

Expected
[image]

imrishabh18, 2024-12-23:
I think so a prop to get set the number of rows, let's confirm with @seveibar

seveibar, 2025-02-22:
/bounty $20

yes `rows={2}` is probably a good way to put it

seveibar, 2025-02-22:
It's actually more of a footprint thing, thought about it a bit more: https://github.com/tscircuit/props/pull/187#pullrequestreview-2634736670