tscircuit/core #446: New trace when added to the already used port having net_label, throws render error

imrishabh18, 2024-12-21:
The motor driver fails to render now, cause that port is already having the net_label and rather that using the same net_label, it just throws render error.

[image]

My hunch is that this bug was introduced in this PR probably - https://github.com/tscircuit/core/pull/422

cc: @Abse2001 @seveibar Should we revert this PR? Meanwhile Abse can take a look at this issue

Abse2001, 2024-12-21:
@imrishabh18 Yeah I know the problem and Im working on a fix

Abse2001, 2024-12-21:
@imrishabh18 The problem is that when the trace complixity change a trace into a label it does not take into account the label rule that no label should render above another label so the error is legit but we should handle this differently.

closed 2025-03-06 by github-actions[bot]