tscircuit/core #462: implement P-Channel MOSFET

ShiboSoftwareDev, 2024-12-28:
I think it's the last component needed for the wifi camera module, very important

[image]

ShiboSoftwareDev, 2024-12-28:
@seveibar what's required to implement this?

seveibar, 2024-12-29:
well first we need to implement the props, technically a `<Mosfet />` component should be an extension of the `<Transistor />` component, but this detail isn't super important right now. The important thing is to get the `props` and `circuit-json` defined.

- [ ] `props` for mosfet
- [ ] `circuit-json` for mosfet
- [ ] symbols for mosfet in schematic-symbols
- [ ] `Mosfet` class in core

seveibar, 2024-12-31:
@ShiboSoftwareDev i was looking into this and it seems there are kind of two properties: `channel_type: "n" | "p"`, `mosfet_mode: "enhancement" | "depletion"`

PR #481 (2025-01-05): implement mosfet
/claim #462
/closes #462
closed 2025-01-05 by ShiboSoftwareDev