tscircuit/core #465: Integrate elkjs for automatic schematic layout

seveibar, 2024-12-29:
https://github.com/kieler/elkjs

CC @ShiboSoftwareDev this is a pretty interesting one but idk how to scope it. It might inflate the bundle too much

ShiboSoftwareDev, 2024-12-29:
@seveibar if the library is big put it in a server and create an api for it like the freerouting api

ShiboSoftwareDev, 2024-12-29:
@seveibar I can integrate it into core and see the effect on bundle size if you want

seveibar, 2024-12-29:
@ShiboSoftwareDev yea let's run the experiment, it could be a big deal if it's good.

Yes we could always make it an API call if it's too big!

/bounty $50

seveibar, 2024-12-29:
@ShiboSoftwareDev if it's slow you'll want to do it in an async effect similar to how the api call for the autorouting works. If you get stuck, no worries, happy to take a look at this sometime later

ShiboSoftwareDev, 2024-12-29:
@seveibar let me have a go at it

ShiboSoftwareDev, 2025-01-20:
@seveibar let's start working on this, please layout a plan

seveibar, 2025-01-20:
have we validate elkjs can actually layout a board properly? I think the thing I'm most concerned about is validating that it can actually do a layout that is nice.

I've seen some screenshots that look ok, but if it's too limited I feel like we'd be better off trying to build out a basic circuit-specific layout algorithm

The bounty definitely doesn't make sense. Maybe we have a repo called `experimental-elkjs-layouts` or something? Takes Circuit JSON lays it out?

closed 2025-04-05 by github-actions[bot]