tscircuit/core #518: `internallyConnectedPorts` prop

seveibar, 2025-01-13:
Indicates that ports are internally connected inside of a chip

Solves the "push button problem" because push buttons have internally connected ports, but schematic representation only has two ports

/bounty $25

UTK-a-RSH, 2025-01-21:
How can I test my implementation regarding this issue can anyone please tell me ?

seveibar, 2025-01-24:
@UTK-a-RSH we're implementing this on stream now!!! https://x.com/seveibar 00:45:00

seveibar, 2025-01-24:
tl;dr; this is the repro the trace doesn't connected properly

```tsx
import { PushButton } from "@tsci/seveibar.push-button"

export default () => (
  <board width="30mm" height="10mm">
    <PushButton
      name="SW1"
      pcbX={10}
      schX={-3}
    />
    <PushButton
      name="SW2"
      pcbX={-10}
      schX={3}
    />
    <trace from=".SW1 .pin1" to=".SW2 .pin3" />
  </board>
)
```

Harihara04sudhan, 2025-02-16:
/attempt #533

closed 2025-02-22 by seveibar