tscircuit/core #521: `autorouter="auto-cloud"` violates subcircuit rules (crosses traces)

seveibar, 2025-01-14:
# CURRENT BEHAVIOR

[image]

# EXPECTED BEHAVIOR

[image]

# REPRODUCTION

```tsx
export default () => (
  <board width="10mm" height="10mm">
    <group subcircuit autorouter="auto-cloud">
      <resistor
        resistance="1k"
        footprint="0402"
        name="R1"
        schX={3}
        pcbX={3}
        pcbY={3}
      />
      <capacitor
        capacitance="1000pF"
        footprint="0402"
        name="C1"
        schX={-3}
        pcbX={-3}
        pcbY={3}
      />
      <trace from=".R1 > .pin1" to="net.NET1" />
      <trace from=".C1 > .pin1" to="net.NET1" />
    </group>
    <group subcircuit autorouter="auto-cloud">
      <resistor
        resistance="1k"
        footprint="0402"
        name="R1"
        schX={3}
        pcbX={3}
        pcbY={-3}
      />
      <capacitor
        capacitance="1000pF"
        footprint="0402"
        name="C1"
        schX={-3}
        pcbX={-3}
        pcbY={-3}
      />
      <trace from=".R1 > .pin1" to="net.NET1" />
      <trace from=".C1 > .pin1" to="net.NET1" />
    </group>
  </board>
)
```

seveibar, 2025-01-14:
/bounty $40

imrishabh18, 2025-01-15:
@seveibar Our `dsn-converter` is now fixed and not violating the subcircuit rules as you can take a look here. But we still have a bug with the hover over traces in the image below (just for this edge case), I looked into it and it seems tricky to resolve and taking too much time, most probably some of the current implementation will have to be modified and I don't want to touch a lot of stuff in there cause currently everything is working fine, and anyways this is a specific edge case where both the subcircuit having the same variable name and not getting connected directly to each other and rather through the `net`. I think we can close this issue for now and create a new issue for this specific bug (can be looked later on). Thoughts?

https://tscircuit.com/editor?&snippet_type=board#data:application/gzip;base64,H4sIAOivh2cAA9WTsU7DMBCG9z7FyVM7NHGAMc5SwchQMcDoOE5j0diWcxZIiHfnKFZSkSKYUCtlyH3+z/7kk/WrdwGh0a2Me4TlCkQFywUAfWXtZGjgxTTYCVbwvmfQabPrMFUVhSi2Cy56GGKtTFDRIMiIjhjqINjn/1rtXWxSnBqCHsxAkVQDfAFplaadn9nIW+fQB2PpQH7Dr6YFK3uKbouJDKp7FG/X7yPwqp6BpyOQjzpKeqnMsU8iSYhz7u/+LLWZSa1nVt/IaS0MUmlog+sFy7YFVJB5YwsG6ASzGrP72weqfmjY/NZQ5oe5XeIMp+s7ryGe8vq/KZb54b1Wi9UHFl7b+NUDAAA=

[image]

closed 2025-01-26 by imrishabh18