tscircuit/core #540: When the component is placed outside the board, It should throw error

imrishabh18, 2025-01-21:
https://snippets.tscircuit.com/editor?&snippet_type=board#data:application/gzip;base64,H4sIAPuPj2cAA42QvQ6CMBRGd57iphMMQlFH6ELiAzC5llJso/1JqdHE+O422AbD5Hi+nJuTXP60xnkY+UTvNw95AS2BPANoBkPdCA85etGiWikEgsuL8AFwIBKcYDk+y9kbtxDAF6lmPGhXFNfJGG+d1OEWH/E+zZqqoPV14pmJc/s6vCNaNqxYxR6jljK5BiPHIsbYnv6qdpvqbpNNnLreUcZhcka1qOxrIFBaqWsE3oSh+xmWi6Za3key4gPg67TLYQEAAA==

[image]

When the components are placed outside the board and are connected, then the snippet should throw error and not draw the traces.

imrishabh18, 2025-01-21:
/bounty 10$

imrishabh18, 2025-01-22:
@techmannih Read the description, it says to THROW an error when the components are placed outside the board and connected, and not this is THROWING an error.

UTK-a-RSH, 2025-01-23:
/attempt  #540

UTK-a-RSH, 2025-01-25:
error: 
Something went wrong installing the "sharp" module

Cannot find module "../build/Release/sharp-linux-x64.node"

I am getting this error while testing. Can anybody tell me how can I fix this

seveibar, 2025-01-25:
Just as a note on this, the error should be a circuit json error, we don't want the rendering process to throw

UTK-a-RSH, 2025-01-26:
@seveibar Please Explain this a bit more

UTK-a-RSH, 2025-01-27:
> Just as a note on this, the error should be a circuit json error, we don't want the rendering process to throw

Now I got what you were to trying to make me understand thanks now I have reached to the point.

Abse2001, 2025-02-24:
/attempt #540

PR #649 (2025-02-24): insert a pcb_trace_error when trace out of board
/claim #540
/closes #540
closed 2025-02-24 by Abse2001