tscircuit/core #578: Fix schematic traces routing through symbols

seveibar, 2025-01-27:
[image]

/bounty $20

Rishikesh63, 2025-01-28:
[image]

Rishikesh63, 2025-01-28:
[image]
I think it's right

seveibar, 2025-02-08:
Here is a reproduction @Rishikesh63 

```tsx
export default () => (
  <board width="10mm" height="10mm">
    <resistor
      resistance="1k"
      footprint="0402"
      name="R1"
      schX={-3}
      pcbX={3}
    />
    <capacitor
      capacitance="1000pF"
      footprint="0402"
      name="C1"
      schX={3}
      pcbX={-3}
    />
    <trace from=".R1 > .pin1" to=".C1 > .pin1" />
  </board>
)
```

seveibar, 2025-02-21:
/bounty $30

rohitbhure65, 2025-02-24:
/attempt #578

ShiboSoftwareDev, 2025-02-27:
/attempt #578

Anshgrover23, 2025-03-25:
yup, the issue is resolved.

Anshgrover23, 2025-03-25:
@techmannih no, the issue is still had see 

[image]

closed 2025-05-04 by seveibar