tscircuit/core #593: Invalid props for trace "unnamed" (improve error message by naming specific invalid props)

seveibar, 2025-02-01:
You may need to fix it in NormalComponent or PrimitiveComponent not sure

[image]

Make sure to have a test with an invalid prop

/bounty $15

ArnavK-09, 2025-02-01:
> You may need to fix it in NormalComponent or PrimitiveComponent not sure
> 
> [image]
> 
> Make sure to have a test with an invalid prop
> 
> @seveibar 

[image]

i think it includes prop name, like maxLength its from the svg side that it doesnt include full errors, i tried to fix it,

is this the problem or anything else in the code??

## https://github.com/tscircuit/svg.tscircuit.com/pull/21

seveibar, 2025-02-02:
@ArnavK-09 you're right. The error message should use ellipsis

closed 2025-02-02 by Anshgrover23