tscircuit/core #624: Trace routing is not correct in schmatic view

Anshgrover23, 2025-02-13:
- Correct trace routing
[image]
- Incorrect trace routing

[image]

### Code for reproduction

```tsx
export default () => (
  <board width="10mm" height="10mm">
    <resistor
      resistance="1k"
      footprint="0402"
      name="R1"
      schX={-3}
      pcbX={3}
    />
    <capacitor
      capacitance="1000pF"
      footprint="0402"
      name="C1"
      schX={3}
      pcbX={-3}
    />
    <trace from=".R1 > .pin1" to=".C1 > .pin1" />
  </board>
)
```

Anshgrover23, 2025-02-13:
@seveibar ????? your opinion

seveibar, 2025-02-17:
The small things i don't care about much right now, it's when it goes through the component that i care a lot. If you have a repro for a trace going through a component feel free to reopen

closed 2025-02-17 by seveibar