tscircuit/core #673: Fix schematic text overlay for chip

imrishabh18, 2025-03-01:
Figure out what are these texts and fix it!

[image]

https://docs.tscircuit.com/footprints/footprinter-strings

/bounty $2

seveibar, 2025-03-03:
There are two issues, one issue is that core is outputting left/right as the "best label" for the schematic component, the other issue is that it's not sizing the output schematic component properly to fit the text

/bounty $8

ShiboSoftwareDev, 2025-05-09:
@seveibar the problem here is that footprints like 0805 have port hints "left" and "right", they don't have a pin label though so no space is assigned for them, but we display the port hint anyway, here is the logic that derives the label from port hints:

[image]

we can't just match out left and right because users might name their pins left or right, what do you say?

seveibar, 2025-05-09:
I feel like we should maybe have special handling for left/right in that port hint fn that shibo found, maybe we could do special handling to not show left/right

closed 2025-05-09 by seveibar