tscircuit/core #693: Board automatically defined with the fixed width and height

imrishabh18, 2025-03-07:
For this circuit, we are not defining the `<board />` but still you can see the `<board />` automatically defined with `width` and `height` which is not covering the whole letter `C` but just some led's which are at the origin (that could be another bug).

https://tscircuit.com/editor?snippet_id=1f6c7483-f1e2-4086-a0aa-1abf0668ca25

[image]

/bounty $10

Husainj, 2025-03-08:
/attempt #693

Rishikesh63, 2025-03-09:
@Husainj  if you run code then board is created

[image]

Anshgrover23, 2025-03-09:
@imrishabh18 the board is creating, that is the right bug but it is covering C letter i think there is no bug for that thing.

closed 2025-03-14 by imrishabh18