tscircuit/core #896: create obstacles for square pad plated holes

ShiboSoftwareDev, 2025-06-09:
cc @techmannih

stephane-alnet, 2025-06-09:
Not sure if this is related, but (in tscircuit 0.0.488 at least) the square pads do not show up in the UI. Compare `footprint="pinrow3_p2.54_nosquareplating"`

[image]

 with `footprint="pinrow3_p2.54"`

[image]

seveibar, 2025-06-09:
@stephane-alnet we think that the main `tscircuit` dep recently stopped publishing and slightly out of date versions have the missing pad. @techmannih and I are investigating

seveibar, 2025-06-09:
I believe this is now patched in v0.0.490

closed 2025-08-24 by github-actions[bot]