tscircuit/core #1236: Repro47 - Greatly improve automatic schematic layout for circuits with symmetric groups

seveibar, 2025-08-31:
[image]

I'm not sure what technique or process to use. The key fix is most likely in the matchpack algo repo: https://github.com/tscircuit/matchpack

/bounty $200

AnasSarkiz, 2025-08-31:
@seveibar @imrishabh18  I've added repro at [matchpack](https://github.com/tscircuit/matchpack)
Also, add improvements for automatic schematic layout for circuits with symmetric groups

```tsx
import { sel, RootCircuit } from "tscircuit"
import type { ChipProps } from "@tscircuit/props"

/**
 * Symmetric Switch Matrix Circuit - Repro47
 * This circuit demonstrates a switch matrix with repeated symmetric patterns
 * that should be laid out in an organized grid but currently aren't optimized
 */

export const SymmetricSwitchMatrix = () => (
  <board width="50mm" height="50mm" routingDisabled>
    <group name="SwitchMatrix">
      {/* Row 1 - First set of symmetric switches */}
      <group name="Row1" schX={-15} schY={10}>
        <capacitor
          name="C1_1"
          footprint="0402"
          capacitance="100nF"
          schX={-6}
          schY={0}
        />
        <resistor
          name="R1_1"
          footprint="0402"
          resistance="10k"
          schX={-3}
          schY={0}
        />
        <chip
          name="SW1_1"
          footprint="sot-23"
          pinLabels={{ pin1: ["IN"], pin2: ["OUT"], pin3: ["GND"] }}
          manufacturerPartNumber="BSS138"
          schX={0}
          schY={0}
        />
        <capacitor
          name="C1_2"
          footprint="0402"
          capacitance="100nF"
          schX={3}
          schY={0}
        />
        <resistor
          name="R1_2"
          footprint="0402"
          resistance="10k"
          schX={6}
          schY={0}
        />
      </group>

      {/* Row 2 - Identical pattern to Row1 */}
      <group name="Row2" schX={-15} schY={5}>
        <capacitor
          name="C2_1"
          footprint="0402"
          capacitance="100nF"
          schX={-6}
          schY={0}
        />
        <resistor
          name="R2_1"
          footprint="0402"
          resistance="10k"
          schX={-3}
[... 152 lines cut ...]
      <trace from="C_PWR1.2" to="net.GND" />
      <trace from="C_PWR2.2" to="net.GND" />
      <trace from="C_PWR3.2" to="net.GND" />
      <trace from="C_PWR4.2" to="net.GND" />

      {/* Ground connections for all switches */}
      <trace from="SW1_1.pin3" to="net.GND" />
      <trace from="SW2_1.pin3" to="net.GND" />
      <trace from="SW3_1.pin3" to="net.GND" />
      <trace from="SW4_1.pin3" to="net.GND" />

      <trace from="CTRL1.pin2" to="net.GND" />
      <trace from="C_CTRL.2" to="net.GND" />
    </group>
  </board>
)

export const getSymmetricSwitchMatrixJson = () => {
  const circuit = new RootCircuit()
  circuit.add(<SymmetricSwitchMatrix />)
  return circuit.getCircuitJson()
}
```

https://github.com/user-attachments/assets/0c883b8a-d8d6-431b-849f-e444d1fcf04f

AnasSarkiz, 2025-08-31:
### This video shows  a demo setup to show before/after symmetric group layouts

https://github.com/user-attachments/assets/481af09f-0499-4ce2-904e-805d40460c80

### cc: @seveibar @imrishabh18. 
If you think this is a good fix, let me create a pr