tscircuit/core #1348: Support a `<subcircuit />` having a single schematic box representation

seveibar, 2025-09-14:
We want to enable subcircuits to have a single schematic box representation, this way you can have e.g. a group of pinheaders representing a single schematic box for an arduino shield.

- [ ] Propose any necessary props to tscircuit/props
- [ ] Introduce any changes required to circuit-json, perhaps something like `schematic_component.is_schematic_group`   
- [ ] Implement in core

As discussed here: https://github.com/tscircuit/common/pull/9#pullrequestreview-3221726590

/bounty $400

seveibar, 2025-09-14:
CC @Abse2001

Abse2001, 2025-09-14:
/attempt #1348

BenraouaneSoufiane, 2025-09-14:
/attempt #1357

Prachi-kushwaha, 2025-10-03:
/attempt #1348

rishi-jat, 2025-10-25:
/attempt

rishi-jat, 2025-10-25:
@seveibar is this issue is still open can I work on this issue ?

closed 2026-01-08 by github-actions[bot]

Eelprofesor14, 2026-11-09:
/attempt #1348